freerouting/freerouting #322: Cannot configure different tracing widths based on required current

dpalic, 2024-07-12:
**Describe the bug**
In this example the bug is well explained: https://electronics.stackexchange.com/q/718243/386022

**To Reproduce**
recreating the schematics, but can also be shared if required

**Expected behavior**
if the traces are configured, it shall be also used accordingly. It is a normal case that you have one voltage rail but different current paths

**Screenshots**
see link

**Environment**
 - OS: windows and ubuntu x64
 - Freerouting version: 1.9.0
 - EDA: KiCad

**Command line arguments**
none used

**Project files**
can be provided if required

**Logs**
none available

GorillaSapiens, 2024-10-06:
(i'll include a screenshot from your stackexchange issue below)

it looks like you want to assign two different widths to the same net.  i don't know of any commercial router that will let you do this.  for example, in EagleCAD, a net gets one and only one netclass, which as far as i know can only be assigned one trace width.

for stuff like this, we'd manually route the stretch from J1.1 to R2, and assign a different netclass for the net on the other side of R2.  then let the autorouter take over.  that or place R2 as close as possible to J1.1

i don't know of any commercial autorouter that does what you're asking for.

[image]

GorillaSapiens, 2024-10-07:
i thought of a workaround.  connect J1.1 to a testpoint.  then that testpoint to R2 and U1.  now you can define net classes for J1.1->TP and TP->R2 different from TP->U1.  problem solved.

closed 2025-02-13 by github-actions[bot]