freerouting/freerouting #326: Roadmap to version 2.0

andrasfuchs, 2024-07-19:
There are a few issues already assigned to version 2.0 and there are a few steps that I plan to complete before the next release.

I copy them here from my personal notes for transparency, but I can't give you deadlines since I have very little to work on Freerouting nowadays. As always, any help is welcome!

- [ ] add RESTful web API + generate OpenAPI documentation
- [ ] add a few JUnit test + GitHub actions to execute them on each build
- [ ] create a docker image manually + GitHub actions to do it automatically for the next release
- [ ] update MVN Repository manually + GitHub actions to do it automatically for the next release
- [ ] test KiCad 8 integration (with macOS Homebrew support)
- [ ] publish API to https://api.freerouting.app
- [ ] documentation - settings in freerouting.json file
- [ ] documentation - RESTful API usage, samples, OpenAPI access
- [ ] documentation - EasyEDA and tscirciut integration
- [ ] documentation - generate charts based on the stats
- [ ] documentation - v2.0 release draft: new features, Java 21, localized texts+translations, GUI, API, freerouting.json, sponsors, stats
- [ ] documentation - new screenshots, update README

seveibar, 2024-07-19:
Very excited for tscircuit + freerouting. A lot more people will be able to build electronics if we reduce/eliminate the need to autoroute for simple boards!!!

dpalic, 2024-07-26:
Hi @andrasfuchs 
I may help you with rest/openapi, docker, junit and mvn
but it would require to get some kickstart from your end.
Is it possible to organize a zoom call with all potential helpers?

andrasfuchs, 2024-07-26:
@dpalic That would be awesome, thank you for considering. 
Please feel free to schedule a convenient time using this link: https://cal.com/fuchs

andrasfuchs, 2024-09-10:
@imrishabh18 It is already possible, but many things are still moving around, so please wait at least a few weeks until I can get to a version that works as intended and I got some testing and documentation done.

I will need to focus on my other responsibilities in the next few days and probably next week, so I can't promise any deadlines, but if you have the capacity to implement the OpenAPI or any other tasks from the above list, that would be great and I would check your PRs as soon as I can.

Ashleycodesseo, 2024-09-18:
hi @andrasfuchs I saw that you need help documenting for the RESTful API portion of this project. I'm looking to get some practice and experience documenting so please add me on and let me know what you need help with. I'd be happy to contribute. Thanks

andrasfuchs, 2024-09-19:
@Ashleycodesseo Fantastic, thank you! 
Please drop me a line to info@freerouting.app with your preferred e-mail address, so that I can add you to Postman workspace where you can see the endpoints we have. There is a documentation section in Postman as well, but I haven't used it yet. I plan to put the documentation both on GitHub and into the code using Java attributes that will be used by the OpenAPI interface.
All suggestions are welcome, the current version of the API is just a preview, so don't hesitate to recommend changes to the endpoints and their parameters.

Ashleycodesseo, 2024-09-19:
@andrasfuchs you got it! I'm working through my projects today and I will message you once you email me and I get started on this. My email is ashleycodesseo@gmail.com Thank you!

andrasfuchs, 2024-09-23:
@imrishabh18 @Ashleycodesseo 
I have just published the API to https://api.freerouting.app and I created environments accordingly in Postman.

[image]

You can change your current testing environment in the top right corner in Postman. `Localhost` can be used when you have Freerouting API running locally, `beta` is for my testing environment, and `production` set the API URL to _api.freerouting.app_.

Feel free to play with them, don't worry, you can't break anything, or if you do, that's a valuable feedback too :) It will be changed in the following days and weeks, and there will be a required authentication for the production access as well.

Let me know if you have any issues, questions or other feedback.

Ashleycodesseo, 2024-09-23:
ok @andrasfuchs I will take a look at it today

andrasfuchs, 2024-09-25:
@Ashleycodesseo I've just created a very basic API documentation here: [API_v1.md](https://github.com/freerouting/freerouting/blob/master/docs/API_v1.md). I hope it helps, and I would be greatful if you could improve it. ;)

closed 2024-11-06 by andrasfuchs