freerouting/freerouting #328: SNAPSHOT 20240729_145400 for Windows will not start

andrew64k, 2024-07-29:
The SNAPSHOT 20240729_145400 Windows install runs as usual. The files are installed. No errors.
Starting Freerouting does nothing. No error. No popup. Just nothing.
It exits so fast I don't even see it load.

No options, just double click on Freerouting.

The previous SNAPSHOT (20240726) version installed and ran correctly.

**Environment**
 - OS: Windows 10 Pro 22H2 64-bit Intel

boromyr, 2024-07-30:
Same issue:
Error: Could not find or load main class freerouting-SNAPSHOT-20240729_145300.jar
Caused by: java.lang.ClassNotFoundException: freerouting-SNAPSHOT-20240729_145300.jar

andrasfuchs, 2024-07-30:
Thanks for the report guys. Our main class and its entry point indeed changed in the last few days, so it might be a compiler misconfiguration issue. It needs to be fixed, so I'll take a look soon (hopefully this week).

andrasfuchs, 2024-07-30:
Could you confirm that it works now on your machines as well? 
Link: https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT

andrew64k, 2024-07-30:
Nope. ```freerouting-SNAPSHOT-20240730_213300-windows-x64``` installs but I get an error when I run it:
```java.lang.NullPointerException: Cannot invoke "app.freerouting.management.SegmentClient.setEnabled(boolean)" because "app.freerouting.management.FRAnalytics.analytics" is null```

andrasfuchs, 2024-07-30:
Alright then, I'll get back to it.
Thanks for the quick reply @andrew64k!

andrasfuchs, 2024-07-31:
I made some updates, could you try it again? Thank you!

andrew64k, 2024-07-31:
Different error this time. Still does not run.
[image]

andrasfuchs, 2024-07-31:
Interesting, I had none of the above errors on my local machine, and I can't reproduce the last one.

Anyway, I have just implemented the missing interface on the RoutingJob class.  Could you check it out again?

boromyr, 2024-07-31:
Now it starts correctly.

andrasfuchs, 2024-07-31:
👍 Thank you guys for your patience and assistance!

closed 2024-07-31 by andrasfuchs

andrew64k, 2024-07-31:
The new version ```freerouting-SNAPSHOT-20240731_123200-windows-x64``` is working for me now on Windows 10.

Thanks!