freerouting/freerouting #332: Autorouter function can not be restarted after finishing

andrew64k, 2024-08-01:
**Describe the bug**
After starting the autorouter process and letting it FINISH (not including optimization)
the autorouter process will not start again. It just continues with via optimization.

**To Reproduce**
Start autorouter (click icon).
Let autoroute finish (not including optimization).
Move or delete some or all routes (with delete all icon).
Start autorouter again (click icon).
Autorouter does not fix the missing routes.

**Expected behavior**
Starting autorouter should check for missing/broken/or non optimal routes and fix them.

**Environment**
Windows 10
Freerouting SNAPSHOT-20240731_123200

andrew64k, 2024-08-01:
Now it's working again.... I'll try to reproduce it...

andrew64k, 2024-08-01:
Now it's doing it again.... I did a bunch of manual routing and then clicked to start the autorouter for the rest of the traces. It just skipped it and moved to via optimization.

If is turn off the passes for fanout/autorouting/optimization and turn it on again, then it works.

andrasfuchs, 2024-08-01:
Thank you @andrew64k, I agree, we should always restart with the autorouter even if it was finished before.