freerouting/freerouting #345: `/enqueue` endpoint giving 500

imrishabh18, 2024-09-12:
**Describe the bug**
Serialisation issue 

```
Caused by: jakarta.json.bind.JsonbException:
Error accessing getter 'getRulesFile' declared in 'class app.freerouting.core.RoutingJob'
```

**To Reproduce**
Steps to reproduce the behavior:
1. Hit the endpint `/enqueue`

**Screenshots**
[image]

andrasfuchs, 2024-09-14:
I can't reproduce the problem, could you give me more details about the issue?

I looked into your PR as well, and I'm not against introducing new DTOs if they are justified, I just want to make sure first this is indeed a general issue with the endpoint.

I use Postman for testing, and it all looks good when I try to enqueue a new job:
[image]

imrishabh18, 2024-09-14:
So, if I want to start the server and then hit the endpoint. I am getting this error on hitting `enqueue`

`Caused by: jakarta.json.bind.JsonbException:
Error accessing getter 'getRulesFile' declared in 'class app.freerouting.core.RoutingJob'

https://github.com/user-attachments/assets/4f4296c2-8f8a-4c23-bf0b-df58c15d7223

Also, is it possible you can share that postman collection please for the endpoints?

andrasfuchs, 2024-09-16:
This screen recoding is pretty cool 👍 Here is mine:

https://github.com/user-attachments/assets/f4948eaf-a592-4b54-910e-3c043388c093
Note: The {sessionId} in the URL is set by a script in Postman. Calling the `changeSettings` endpoint is optional. Uploading the DSN input is required before calling `start` and it must be in base64 format. All of this is subject to change, but it works like this for now.

Btw, you can change the API server enabled property by changing the %temp%\freerouting\freerouting.json file accordingly.

Sure, I'll invite you to the Postman workspace, please drop me a line with your e-mail address to info@freerouting.app. I'm curious if you can that getRulesFile issue.

closed 2024-11-06 by andrasfuchs