freerouting/freerouting #348: NoSuchElementException when starting auto-router

fabianlefebre, 2024-09-19:
**Describe the bug**
I run into a Java Error NoSuchElementException when trying to auto-route.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the snapshot `freerouting-SNAPSHOT-20240910_112300-macos-x64.dmg` from https://github.com/freerouting/freerouting/releases/tag/SNAPSHOT
2. Start freerouting
3. Click on "File" -> "Open"
4. Open a .dsn file exported from EasyEDA
5. Click on the button to "Start the auto-router and route optimizer"
6. Error "java.util.NoSuchElementException: No value present" appears directly

**Expected behavior**
Auto routing starts and completes.

**Screenshots**
[image]

**Environment**
 - OS: MacOS 14.6.1
 - Freerouting version: 2.0 Snapshot 20240910_112300
 - EDA: EasyEDA
 - Java: java version "21.0.4" 2024-07-16 LTS

**Project files**
Not applicable. 

**Logs**
```
INFO    Freerouting v2.0.0-SNAPSHOT (build-date: 2024-09-10)
INFO    Settings were loaded from freerouting.json
WARNING Failed to check for new version: java/net/http/HttpClient
ERROR   Failed to read file contents.
ERROR   No value present
ERROR   No value present
```

**Additional context**
Please contact me for the file, as I cannot publish that schematic. 
The same file routes very well under version 1.9

andrasfuchs, 2024-09-20:
Thanks for the report @fabianlefebre!

Could you please send your input file to info@freerouting.app? I could test it early next week.

andrasfuchs, 2024-09-25:
It was an issue with the routing job scheduler, it should be ok now.

closed 2024-09-25 by andrasfuchs