freerouting/freerouting #351: routes through NPTH

GorillaSapiens, 2024-09-28:
looks like freerouting is perfectly happy to run a route right through a non plated through hole for mechanical mounts.

8 layer board.  route goes right through a NPTH

kicad 8.0.5, freerouting plugin 1.9

[image]

andrasfuchs, 2024-10-01:
@GorillaSapiens Thank you for the report, it looks like an important bug.

Could you share the DSN file with me, so that I can reproduce it?

If that's not possible, could you try it again, without the optimization step? You can turn that off in the auto-router settings:
[image]

If there is still an issue, could you reduce the number of layers to 6, then to 4, and then to 2 and try again?

Thank you!

GorillaSapiens, 2024-10-02:
looks like github won't let me attach a DSN here.  is there an email i can send it to?  the bug is trivial to reproduce.  I used two 1206 resistors on a 4 layer board, placed via in pads, and a large oval NPTH in the middle.
[image]

GorillaSapiens, 2024-10-02:
found your email and sent you the DSN.  subject is "DSN file for Freerouting github issue #351"

andrasfuchs, 2024-10-02:
@GorillaSapiens I've just got it, thanks a lot! 👍 I'll look into it!

GorillaSapiens, 2024-10-02:
also, i'm not sure if this is related or not, it seems like Freerouting is ignoring anything i put on the "margin" layer in Kicad.  if i put down a circle on the "margin" layer, i expect no routes there, and no vias there.

closed 2025-02-07 by github-actions[bot]