freerouting/freerouting #356: unconnected islands in fill zones

GorillaSapiens, 2024-10-07:
kicad 8.0.5, freerouting plugin 1.9.0

freerouting literally leaves unconnected islands in fill zones that could be easily connect.

given this and other bugs, i cannot reasonably expect that FreeRouting's solution is correct.

i'm back to hand routing this board.  DEEPLY disappointed.

[image]

GorillaSapiens, 2024-10-07:
[image]

GorillaSapiens, 2024-10-07:
this has more to do with fill settings in kicad.  this isn't a freerouting bug.  sorry.  closing.

closed 2024-10-07 by GorillaSapiens

andrasfuchs, 2024-10-10:
No worries, thank you for letting me know!