freerouting/freerouting #359: Always route optimize in signle thread

L1uTongweiNewAccount, 2024-10-16:
In version 1.9.0, If you not set `-mt` option, it always run routing optimizition with signle thread.

You can have a repetition in Windows 10, with:

`"C:\Program Files\Common Files\Oracle\Java\javapath\java.exe" -jar C:\Users\homeuser\AppData\Local\freerouting\app\freerouting-executable.jar`

My Java is 22.0.1!

L1uTongweiNewAccount, 2024-10-16:
I know is a bug, because the home page say if no `-mt` option, it should be set as `cpu_cores - 1`.

andrasfuchs, 2025-02-20:
I aim to fix this in the v2.1 release. I need to test it thoroughly though, because the multi-threaded optimizer caused a lot of clearance violations in the past.

L1uTongweiNewAccount, 2025-02-20:
> I aim to fix this in the v2.1 release. I need to test it thoroughly though, because the multi-threaded optimizer caused a lot of clearance violations in the past.

OK. For sure, it's also important to add multiple threads to router.