freerouting/freerouting #360: Freerouting plugin stopped working

dp1044, 2024-10-16:
**Describe the bug**
Yesterday morning I finished a project and freerouting worked fine.  Yesterday Afternoon it stopped working.
Freerouting plugin for kicad 7 and 8 does not complete routing all the traces.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to kicad 7/8
2. Click on Freerouting plugin
3. click on autoroute
4. It routes only a few then quits back to kicad with only those routes connected

**Expected behavior**
To have freerouting finish all traces before if finishes

**Screenshots**
Here are shots of the board before and after autorouting.
[image]
[image]

**Environment**
 - OS: Windows 11, 64-bit
 - Freerouting version: 1.8 and 1.9
 - EDA: Kicad 7 and 8.0.6

**Command line arguments**
N/A

**Project files**
zip of the project
[BattMonitor.zip](https://github.com/user-attachments/files/17403570/BattMonitor.zip)

**Logs**
no errors or logs

**Additional context**
I have tried the other suggestions, uninstall and reinstall of the plugin, kicad and jre. Nothing was changed on my setup or installed between it working and stopping.

boromyr, 2024-11-16:
same issue with version 2

boromyr, 2024-12-25:
Does anyone have this problem? For me FR no longer works completely, it routes 5% of the routes and then closes.

anomaly256, 2024-12-26:
Yes I am having exactly the same issue.  A few traces created, everything else 'failed' but no reason given.

[image]

anomaly256, 2024-12-26:
@boromyr actually, for my case, I've found the culprit - the footprints for the sparkfun/adafruit PCBs I'm using have the rectangular outline on the edgecuts layer, causing freerouting and kicad itself to fail to connect routes to those pads.  Moving the outline to the silkscreen layer fixed the auto routing.

I'm not sure if U1 in your case is a custom footprint or not but it might be worth checking which layer things are on and making sure they're correct, or making a new custom footprint.

boromyr, 2024-12-26:
[image]
In my case everything is partially routed, it's not a problem with the single footprint. Where could I find more specific log files?

closed 2025-05-03 by github-actions[bot]