freerouting/freerouting #367: Routing failed

coliss86, 2024-11-03:
**Describe the bug**
Hello,

I'm new to kicad and freerouting, forgive me if I'm doing something wrong.
I'm experiencing a bug when auto routing from kicad: one track is missing.
 
**To Reproduce**
I have attached my whole project to help to reproduce the problem.
1. Open the file UltraFlactyl.kicad_pro in Kicad
2. Go to PCB view
3. Clear all route: Edit menu > Global Deletions... > Select Tracks and via > OK
4. Launch Freerouting with the button in the button bar
5. Wait for the routing to finish

**Expected behavior**
All track should be routed

**Screenshots**
Here is a screenshot of my problem.
[image]
DRC:
[image]

**Environment**
 - OS: macos 14.7
 - Java: OpenJDK Runtime Environment Temurin-23.0.1+11 (build 23.0.1+11)
 - Freerouting version: 1.9.0
 - EDA: KiCad 8.0.2-1

**Command line arguments**
If applicable, include the full command line that you used.

**Project files**
[Archive.zip](https://github.com/user-attachments/files/17611280/Archive.zip)
It needs symbol from: 
* Blackpill https://github.com/donys8/kicad-lib-WeAct-Black-Pill-V2.git
* scottokeebs https://github.com/joe-scotto/scottokeebs/tree/main/Extras/ScottoKicad

**Logs**
Include at lest the most recent lines of your log files or console output.

**Additional context**
Add any other context about the problem here.

Feifei-0206, 2024-11-08:
我PCB逃逸布线的新手，大佬能带带吗

andrew64k, 2024-11-16:
I have the same problem with the current version (2.0.2 snapshot). Autoroute completes but with 4 connections not found (as listed in the bottom bar). They are simple connections that are just not finished. Earlier freerouting versions can complete all the routes.
[image]

boromyr, 2024-11-23:
same problem

jsiddall, 2025-01-20:
I have seen this issue a few times also

L1uTongweiNewAccount, 2025-02-19:
same problem, and maybe like issue #424, In some situations, router will think some wires (usually spacing is similar with the max spacing or at the via) are impossible.

L1uTongweiNewAccount, 2025-02-20:
@andrasfuchs @coliss86 I found a new key to the problem. When you set the speed option to "slow" instead to "fast", this situation may better.

If the "fast" mode use some kinds of wrong optimization that ignores some wires? I can say the rate of mistaken of fast mode was much more than slow mode, especially the board is very complex.

It always left 3-5 wires not connected in my project, but it does well in slow mode.

More importantly, fast mode is a default option. Many  people will think the router is bad because it cannot solve easy wires in fast mode.

Anyway, please test more about fast mode, and see the problems.

boromyr, 2025-02-20:
I tried the slow mode but nothing changed for me.

L1uTongweiNewAccount, 2025-02-20:
> I tried the slow mode but nothing changed for me.

Sorry to hear that, but slow mode really works for me.

So I also think the change of new algorithm has some wrong optimizion...

andrew64k, 2025-02-22:
***Slow*** did help for me (routes complete). Maybe the default should be Slow. If the ***Fast*** algorithm can't be fixed then it should automatically change to Slow to try to complete the routing, or at least give a clickable notice to try the Slow option.

L1uTongweiNewAccount, 2025-02-23:
@andrew64k Right. I am sure there are some wrong optimizion in fast mode. @andrasfuchs

andrew64k, 2025-02-25:
Slow is still not 100%. I ran into a different, but still missed/incomplete easy connection.

andrasfuchs, 2025-02-25:
Could you guys give me a DSN file/KiCad project that fails in slow mode as well?

boromyr, 2025-03-02:
[Charger (2).zip](https://github.com/user-attachments/files/19043290/Charger.2.zip)

andrasfuchs, 2025-03-07:
I spent 4 hours on this issue today, and although I made progress, it's still fails to complete the routing.
I made the logging more verbose to have a bigger chance caching the real issue here.

```
2025-03-07 15:39:09.356 INFO   [9427B3\CF4A7C] Starting routing of 'UltraFlactyl'...
2025-03-07 15:39:09.648 WARN   The normalization of net 'Net-(D28-A)' failed.
2025-03-07 15:39:09.730 WARN   The normalization of net 'Net-(D33-A)' failed.
2025-03-07 15:39:14.892 DEBUG  [9427B3\CF4A7C] Autorouter Failed to route connection between pin of component #36 and pin of component #66, because the maze search algorithm could not be created.
2025-03-07 15:39:15.737 DEBUG  [9427B3\CF4A7C] Autorouter Failed to route connection between pin of component #66 and pin of component #36, because no connection was found between their nets.
2025-03-07 15:39:16.103 INFO   [9427B3\CF4A7C] Auto-router pass #1 on board 'cb2c292d310f584e3114495c98f0a52c' was completed in 6.57 seconds.
2025-03-07 15:39:17.632 DEBUG  [9427B3\CF4A7C] Autorouter Failed to route connection between pin of component #36 and pin of component #66, because the maze search algorithm could not be created.
2025-03-07 15:39:18.239 DEBUG  [9427B3\CF4A7C] Autorouter Failed to route connection between pin of component #66 and pin of component #36, because the maze search algorithm could not be created.
2025-03-07 15:39:18.277 INFO   [9427B3\CF4A7C] Auto-router pass #2 on board '41f379a56757bffd3ec697ce255685f1' was completed in 2.14 seconds.
2025-03-07 15:39:18.868 DEBUG  [9427B3\CF4A7C] Autorouter Failed to route connection between pin of component #36 and pin of component #66, because the maze search algorithm could not be created.
2025-03-07 15:39:19.454 DEBUG  [9427B3\CF4A7C] Autorouter Failed to route connection between pin of component #66 and pin of component #36, because the maze search algorithm could not be created.
2025-03-07 15:39:19.487 INFO   [9427B3\CF4A7C] Auto-router pass #3 on board 'b3d62d46c4648bb8b58961f7d2c0a9ed' was completed in 1.18 seconds.
2025-03-07 15:39:19.497 INFO   [9427B3\CF4A7C] Auto-routing was completed in 10.14 seconds.

```

The normalization warning should not be a problem. The pins are part of the net "Net-(D34-A)", but the maze algorithm can't create "doors" between the "rooms" and it fails at creation. _Rooms_ are the representations of component shapes and free spaces.

Anyway, I have to stop now, but I will try to fix this one more time next week.

andrasfuchs, 2025-03-10:
As I investigation the issue further for a few more hours I realized that what the slow mode does that it disables the optimization of the ShapeSearchTree class for the 90 and 45-degree restrictions.

I'll run a few speed tests as well to see how much we gain by keeping the fast mode later. (Fast mode makes the codebase significantly more complex, so I would be glad to discard it, if it can be justified.)

I planned to run 5 different boards with default settings, in all modes, both the router and optimizer enabled, multi-threading disabled. As I started testing I found an issue that needs to be dealt with first because my first test file made the router use up much more memory than it should in fast-free and slow-45 modes.

I'll update the results here:
| File Index 	| File Size 	| To route 	| Failed 	| Algo      	| Router 	| Optimizer 	|
|------------	|-----------	|----------	|--------	|-----------	|--------	|-----------	|
| 1          	| 8k        	| 33       	| 0      	| fast-45   	| 2.45s  	| 3.84s     	|
| 1          	| 8k        	| 33       	| 14     	| fast-90   	| 0.76s  	| 0.28s     	|
| 1          	| 8k        	| 33       	| N/A    	| fast-free 	| N/A    	| N/A       	|
| 1          	| 8k        	| 33       	| N/A    	| slow-45   	| N/A    	| N/A       	|

andrasfuchs, 2025-03-10:
I agree, but the slow mode fails in many cases (probably related to the memory leaks issues #390 #420) at the moment. I'd like to fix that first, then measure the difference between fast and slow modes, and if that's not significant then remove the fast mode.

andrasfuchs, 2025-03-12:
After conducting the above tests, I arrived at these conclusions: 
1. we keep the slow mode, because sometimes it solves failed routings
2. fast mode is significantly faster in some cases
3. a hybrid approach would be great: do the first pass in fast mode, then switch to slow mode, then try fast again, etc.
4. keeping the mode selection on the UI wouldn't make too much sense if the hybrid mode works
5. repeated failed routing attempts can slow down the whole process, so one optimization effort could be to reduce them

Implementing hybrid mode isn't trivial, because I might need to maintain the supporting structures in memory for both algorithms at the same time, and there would be an extra overhead maintaining them for both modes. Memory consumption is already an issue for bigger boards.

jsiddall, 2025-03-12:
Interesting results.

I will add that, at least in some of the cases I have experienced where routing finished with failed traces, there was only one trace missing.  In the case I am thinking of it was a very simple connection -- nothing that required ripping anything up to make it fit.  This seems like it may be a different problem than the example you showed above where 94 failed.

If it would be helpful I could also try re-routing with an older version because I am pretty sure I was able to complete the same (or a similarly complex board) with and older version.  Maybe that would help identify the location of the problematic code.

andrasfuchs, 2025-03-28:
@jsiddall Sorry for the late reply, I had to focus on another project in the last 2 weeks.

So, yes, I agree, having a missing route at the end seems to be a recurring problem, and needs to be dealt with.

I implemented an (experimental) scoring system today, that would give a score to a board based on its metrics, including a significant penalty for unrouted nets. I'll work on the algorithm using this scoring system, and we'll see how it pans out. I'll keep you updated as I can spare some time to work on it.

Thank you for your kind offer, I'll need to go back to older versions as well and compare the results anyway, so there is no need to do extra testing on your part.

I'm not satisfied with the current performance of the router at all, so I really hope I can make a meaningful improvement in the next version.

andrasfuchs, 2025-03-28:
This table is just for my record keeping. I'll test older versions of Freerouting and also the new ones I'm working on. The new ones will be the v2.1x where x is a letter representing the revision. v2.1 doesn't have slow and fast modes. I use the default settings, 45-degree mode, optimization turned off. 
(I'm not sure I'm able to do it, but I'll try to apply the new scoring system retroactively for the older versions as well.)

| File Index 	| File Size 	| To route 	|         Algo 	|                 Score 	| Router 	|
|-----------:	|----------:	|---------:	|-------------:	|----------------------:	|-------:	|
|          1 	|        8k 	|       33 	| v2.0-fast-45 	|      N/A (0 unrouted) 	|  2.45s 	|
|          1 	|        8k 	|       33 	|     v2.1a-45 	|   911.78 (1 unrouted) 	|  3.48s 	|
|          2 	|       24k 	|       23 	| v2.0-fast-45 	|      N/A (2 unrouted) 	|  0.63s 	|
|          2 	|       24k 	|       23 	|     v2.1b-45 	|   724.95 (2 unrouted) 	|  0.67s 	|
|          3 	|       43k 	|      702 	| v2.0-fast-45 	|                   N/A 	|    N/A 	|
|          3 	|       43k 	|      702 	|     v2.1a-45 	|    N/A (108 unrouted) 	|  46.5m 	|
|          3 	|       43k 	|      702 	|     v2.1b-45 	| 376.45 (122 unrouted) 	|  22.9m 	|
|          4 	|       50k 	|      107 	|     v2.1a-45 	|      N/A (9 unrouted) 	|   4.6m 	|
|          4 	|       50k 	|      107 	|     v2.1b-45 	|  730.03 (17 unrouted) 	| 78.72s 	|
|          5 	|       36k 	|       64 	| v2.0-fast-45 	|      N/A (0 unrouted) 	|  3.47s 	|
|          5 	|       36k 	|       64 	|     v2.1b-45 	|   899.62 (0 unrouted) 	|  4.66s 	|

andrew64k, 2025-03-31:
SNAPSHOT-20250331_182400 I working better for me (now completes routing, where it failed before).

andrasfuchs, 2025-04-02:
I made some further improvement to the router, and now it routes boards that it couldn't do before.
Also, I removed the slow/fast option, now they are integrated, making the default algorithm more robust, without much overhead.
I still have some boards that it can't route (including the one in this issue), and the process got longer with some boards because it tries harder to solve the board, so I'm not totally satisfied. 
I can't spend more hours on this issue now, so we will have this better-but-not-perfect version in the next release.

adamlamine, 2025-04-12:
As a workaround reducing via cost in the autoroute settings seemed to help with getting the last 1-2 connections.

CaptainDangeax, 2025-05-05:
Same issue. circuit completed with 1.4.4, not completed with 2.0.1. Just to follow