freerouting/freerouting #369: Version 2.0.0 won't start on Windows.

andrew64k, 2024-11-07:
Thanks for version 2.0!!! I see you (and others) put in a lot of work. 

**Describe the bug**
After installing the new 2.0.0 version ```freerouting-2.0.0-windows-x64``` on Windows 10 (22H2) successfully. Running it starts but fails with the error: ```java.lang.NoClassDefFoundError: java/util/logging/Logger``` and then ```Failed to launch JVM```

**To Reproduce**
Install on Windows.
Start Freerouting.

**Screenshots**
[image]

**Environment**
 - OS: Windows 10 (22H2) 64 bit
 - Freerouting 2.0.0

Free Routing 1.9 works correctly.
SNAPSHOTS up to and including ```freerouting-SNAPSHOT-20241030_140700-windows-x64``` work correctly.

After that version it does not run on Windows.

andrew64k, 2024-11-07:
```freerouting-SNAPSHOT-20241031_090700-windows-x64``` installs, but won't run. I noticed the install file went from 46MB to 92MB.

andrew64k, 2024-11-07:
I **don't** have a current version of Java installed on Windows. I'm using the version packed with the Freerouting install.

mikhailmihalkov, 2024-11-07:
java update to version 21 does not help, startup error remains

andrasfuchs, 2024-11-07:
Thank you @andrew64k and @mikhailmihalkov!

I can confirm that something is wrong with the included Java JRE in the installer, probably a Java library is missing.

Could you try to go to `c:\Users\<username>\AppData\Local\freerouting\app\` and execute the `freerouting-executable.jar` directly? It should run if you have a JRE 21+ installed.

mikhailmihalkov, 2024-11-07:
> AppData
In this version, Win 10 64 is launched

mikhailmihalkov, 2024-11-07:
PCM planned for release kicad ?

cyboff, 2024-11-07:
> Could you try to go to `c:\Users\<username>\AppData\Local\freerouting\app\` and execute the `freerouting-executable.jar` directly? It should run if you have a JRE 21+ installed.

It works when launched this way with JRE installed from OpenJDK21U-jre_x64_windows_hotspot_21.0.5_11.msi

closed 2024-11-11 by andrasfuchs

andrew64k, 2024-11-11:
It's different now, but still failed... ```freerouting-SNAPSHOT-20241111_140100-windows-x64``` **Exception Occurred: java.lang.RuntimeException: Failed to create BigQuery client**
[image]

andrasfuchs, 2024-11-12:
Thank you Andrew for the quick reaction! 
I'm working on it, it will be fixed soon in v2.0.1.

andrew64k, 2024-11-13:
```freerouting-SNAPSHOT-20241112_222300-windows-x64``` runs for me on Windows 10. Thanks!

commit 753693ea (2024-11-11): Add a few Java modules to the installers (fixes #369)