freerouting/freerouting #375: connection not found

golkit1, 2024-11-11:
I'm using Macos 14.5 M1 pro. 
I downloaded v2.0 and run it on a board I made.
I get this error:
Autoroute complete. 208 connection not found. 
What it can be?
[image]

andrew64k, 2024-11-14:
I see the same problem after the optimization finished. It says a large number of connections not found.
Most of the time it did complete all of the connections. Sometimes it does not, but only a few are left unconnected.

andrasfuchs, 2024-11-14:
Thanks @golkit1 for the report, and @andrew64k for the reaction!
That was a small carelessness on my part, it should display the correct number now. You will see the changes in the next release.

closed 2024-11-14 by andrasfuchs

andrew64k, 2024-11-14:
It's better now. But the autoroute now seems to fail on the last few connections but starts the optimizer anyway (if selected). Before it was finding all the connections and reported the vias before starting the optimizer.

commit 538774c4 (2024-11-14): Update the ratsnest before displaying the auto-router finished message (fixes #375)