freerouting/freerouting #382: Feature Request: While routing, show which nets are "failed"

DeflateAwning, 2024-11-18:
**Is your feature request related to a problem? Please describe.**
When using Freerouting on a very tight PCB, it can be tough to tell if I need to adjust the PCB layout in order to making routing easier/possible. 

**Describe the solution you'd like**

A pane on the right side which shows the names of the net(s) that are currently counting toward the "Failed" stat at the bottom of the screen.

A bonus version of this feature would keep a running tally of those nets, and rank them based on how many passes/times they've appeared in that bad list, so you can see which ones are frequently causing issue (e.g., in a case where two nets fight over a space only big enough for a single trace). 

**Describe alternatives you've considered**

Currently, to review these cases, you must either:

1. Wait for the autorouter to finish (with failed routes)
2. Manually end it half-way through and reviewing which nets aren't routed, then try again

**Additional context**

I use Freerouting via the KiCAD plugin.

jsiddall, 2025-01-20:
+1, this would be a great feature

DeflateAwning, 2025-05-21:
This is still relevant!