freerouting/freerouting #388: Get in-progress traces using API

seveibar, 2024-12-02:
**Is your feature request related to a problem? Please describe.**

I'd like a way to get traces from the API in realtime so that https://tscircuit.com can show the autorouting working

**Describe the solution you'd like**

Possibly just the ability to get the job output early (the dsn session), currently it throws...

```
{ error: "The job hasn't finished yet." }
```

algora-pbc, 2024-12-03:
💎 **tscircuit** is offering a **$50** bounty for this issue
👉 Got a pull request resolving this? Claim the bounty by adding `@algora-pbc /claim #388` in the PR body and joining algora.io

Ujj1225, 2024-12-08:
@seveibar 
Here, I assume you want to know in realtime how the routing is progressing?
Can i just send some of the output (like partial data) during the autorouting process, instead of waiting until everything is finished?

Ujj1225, 2024-12-08:
I saw your speculative fix. Can i use Running state to approach this?
Something like this:

if job state is RUNNING or COMPLETED:   // I saw in routing job state file about the running state
    start streaming output to client // We can use outputStream.write to get partialProgress.getBytes() and flush it when done

    while job is RUNNING:
        send progress message "in-progress"
        wait for 2 seconds // or any amount of time we can use sleep for a certain amount of time

    if job is COMPLETED:
        send final message "completed" with output data

andrasfuchs, 2024-12-08:
I might be mistaken, but I think we already have this implemented. 

Did you try to poll the [GET /jobs/{jobId}](https://github.com/freerouting/freerouting/blob/master/docs/API_v1.md#progress-reports) endpoint while the state of the job is `RUNNING`? 
It should return an JSON that has an output object containing the work-in-progress state of the board.

seveibar, 2024-12-09:
@andrasfuchs here's an example of the payload I get when I do `/v1/jobs/{job_id}` with a running job- this is on my local docker build so apologies if something is out of date!!

I don't see any work in progress state of the board?

To be clear, I'm looking for traces to display in the tscircuit preview so the user can visually see routing like in the freerouting app.

```
{
  id: '59018016-e367-440f-aece-012b2f50f14e',
  created_at: '2024-12-09T09:41:31.448657920Z',
  started_at: '2024-12-09T09:42:00.253046815Z',
  finished_at: null,
  input: {
    size: 132032,
    crc32: 1107893315,
    format: 'DSN',
    statistics: {
      host: "KiCad's Pcbnew,(5.1.9)-1",
      layer_count: 4,
      component_count: 54,
      netclass_count: 1,
      total_net_count: 288,
      unrouted_net_count: null,
      routed_net_count: 0,
      routed_net_length: null,
      clearance_violation_count: null,
      via_count: 43
    },
    filename: 'untitled.dsn',
    path: ''
  },
  snapshot: null,
  output: null,
  session_id: '9998695e-963c-4281-b69d-04da85da92fc',
  name: 'untitled',
  state: 'RUNNING',
  priority: 'NORMAL',
  stage: 'ROUTING',
  router_settings: {
    default_preferred_direction_trace_cost: 1,
    default_undesired_direction_trace_cost: 2.5,
    job_timeout: '12:00:00',
    max_passes: 100,
    fanout_max_passes: 20,
    max_threads: 1,
    improvement_threshold: 0.01,
    trace_pull_tight_accuracy: 500,
    allowed_via_types: true,
    via_costs: 50,
    plane_via_costs: 5,
    start_ripup_costs: 100,
    automatic_neckdown: true
  }
}
```

andrasfuchs, 2024-12-09:
Ah, alright, that's something more detailed than what we have now. 

The currently expected behavior is that the output field in the payload changes from null to an object containing similar fields to the input part. Those values should change every time a new snapshot is created in the backend which is typically after each pass.

You would need something more complete, including the current state of the routing (probably in Specctra SES format?).

seveibar, 2024-12-09:
@andrasfuchs yep! I love how the freerouting app gradually shows you updates (I think for each snapshot?), it gives a great sense of the progress of the board!!

The SES format is great! Although happy to have any format that can be converted into traces (we convert everything into ["circuit json"](https://github.com/tscircuit/circuit-json) at the end of the day) So JSON etc. is fine.

Good to know that output is supposed to be populated- maybe I grabbed the json at the wrong moment, I did find the `output` field was populated for this design in the completed state- so I might have just grabbed it at the wrong point in time. Still I think the main idea of the issue is to have the WIP or speculative traces!

```json
// ...
  "stage": "IDLE",
  "state": "COMPLETED",
  "output": {
    "path": "",
    "size": 0,
    "crc32": 0,
    "format": "SES",
    "filename": "tscircuit-f7271861-1fec-4dd0-a57b-865de75e389e.ses",
    "statistics": {
      "host": "KiCad's Pcbnew,",
      "via_count": 1,
      "layer_count": 2,
      "netclass_count": null,
      "component_count": 13,
      "total_net_count": null,
      "routed_net_count": 18,
      "routed_net_length": null,
      "unrouted_net_count": null,
      "clearance_violation_count": null
    }
  },
// ...
```

andrasfuchs, 2025-01-15:
The goal would be to include the updated board in [the progress update API call](https://github.com/freerouting/freerouting/blob/master/docs/API_v1.md#progress-reports), I think.

This would mean several changes has to be done in the backend. 

I would first look into the mechanism we have now for updating the GUI while the routing is running. If I remember correctly there is a algorithm that limits the refresh rate of the UI to ~10 fps. Something similar could be done, firing an event every second or so (maybe on the NamedAlgorithm class?) to update the current property of the running job if new routes were done to the board.

The handler of that event would sample the current state of the board and converted it into the SES (using the BoardManager's saveAsSpecctraSessionSes method) that we could make available to download through the API mentioned above after we set the Output or Snapshot (?) field of the RoutingJob class.

That would be my high level plan. :)

At which point are you stuck at the moment?

closed 2025-04-12 by andrasfuchs