freerouting/freerouting #399: Routing ignores clearance to TH pads (EasyEDA PRO)

handmade0octopus, 2025-01-05:
**Describe the bug**
Router ignores clearance for TH pads and just runs through them (also apparent in 1.9 version)

**To Reproduce**
Route file attached

**Expected behavior**
Do not route through middle of TH.

**Screenshots**
[image]
[image]

**Environment**
 - OS: W11 64bit
 - Freerouting version: 2.0.1
 - EDA: EasyEDA Pro

**Command line arguments**
Tried ./freerouting.exe -mt 1 but it seems to ignore it completely

**Project files**
https://www.dropbox.com/scl/fi/gmkrjkqe8ku26boztzpka/dsn-error.dsn?rlkey=3yd9091hzpd3n1w2o9zijf2xd&dl=0

**Logs**
```
INFO    Freerouting v2.0.1 (build-date: 2024-11-14)
INFO    Settings were loaded from freerouting.json
WARNING Multi-threaded route optimization is broken and it is known to generate clearance violations. It is highly recommended to use the single-threaded route optimization instead by setting the number of threads to 1 with the '-mt 1' command line argument.
INFO    Starting auto-routing...
INFO    Auto-router pass #1 on board 'b8558861961628f3f2dab03825a5525a' was completed in 4.41 seconds
INFO    Auto-router pass #2 on board 'bed1c401e9a3cdee6c4a043db1105622' was completed in 2.64 seconds
INFO    Auto-router pass #3 on board '0f2fbf9d9b64f8a3f06d88bd0af48728' was completed in 1.50 seconds
INFO    Auto-router pass #4 on board '49ecd3babb581f168b5c9e2ce269b6db' was completed in 0.15 seconds
INFO    Auto-router pass #5 on board '60cfa8f50ff123c989abe648a6482563' was completed in 0.67 seconds
INFO    Auto-router pass #6 on board '3eabaa8b21d0bfd42b83be87b21839b8' was completed in 0.04 seconds
INFO    Auto-router pass #7 on board '76634a23ffcd7948ddf1e1ac3be10707' was completed in 0.66 seconds
WARNING There were only 17.60 track length increase in the last 5 passes, so it's very likely that autorouter can't improve the result further.
INFO    Auto-router pass #8 on board '2b20dd8a89703229cf80fd925089b72f' was completed in 0.02 seconds
INFO    Auto-routing was completed in 10.40 seconds.
INFO    Multi-threading is disabled in the settings. Using single thread.
INFO    Starting route optimization on 1 thread...
WARNING OptViaAlgo.opt_via_location: move via failed
ERROR   Error during autoroute_item
WARNING OptViaAlgo.opt_via_location: move via failed
ERROR   Error during autoroute_item
WARNING OptViaAlgo.opt_via_location: move via failed
WARNING OptViaAlgo.opt_via_location: move via failed
WARNING OptViaAlgo.opt_via_location: move via failed
ERROR   Error during autoroute_item
WARNING OptViaAlgo.opt_via_location: move via failed
WARNING OptViaAlgo.opt_via_location: move via failed
WARNING OptViaAlgo.opt_via_location: move via failed
ERROR   Error during autoroute_item
WARNING OptViaAlgo.opt_via_location: move via failed
WARNING OptViaAlgo.opt_via_location: move via failed
WARNING OptViaAlgo.opt_via_location: move via failed
WARNING OptViaAlgo.opt_via_location: move via failed
WARNING OptViaAlgo.opt_via_location: move via failed
WARNING OptViaAlgo.opt_via_location: move via failed
INFO    Route optimization was completed in 2 minute(s) 10.71 seconds and it improved the design by ~8.04%.
```

handmade0octopus, 2025-02-06:
@easyeda2021 is there a way to add it manually or default it to some value?

Gifford47, 2025-03-31:
you can customize your .dsn file with addiotional rules. f.e.:
```
    (grid via   0.25)
    (grid wire   0.25)
    (grid place   0.25)
    (rule(clear 6))
    (rule(clear 6 (type default_smd)))
    (rule(clear 6 (type smd_smd wire_area via_smd via_pin pin_pin track_pin)))
    (rule(clear 6.3 (type via_via)))
```

Keep aware of actual selected unit within freerouting. The selected unit must be consitent with the unit in the .dsn file (mostly "mil" in easyeda).
you should check the top of every dsn file, especially the "rule(...)" sections. the easyeda dsn exporter has several bugs ... :/

Gifford47, 2025-04-08:
hi @handmade0octopus @easyeda2021 

please see https://github.com/freerouting/freerouting/issues/445#issuecomment-2778586282

there's a lot to do (bug fixing) with the current easyeda export tool ...

handmade0octopus, 2025-04-08:
Thanks @Gifford47 for your work. We will try to pressure easyeda to fix it.

closed 2025-04-08 by handmade0octopus