freerouting/freerouting #413: Freerouting leaves nets unrouted in 90 degree mode

edwardcwang, 2025-02-07:
**Describe the bug**
Freerouting leaves nets unrouted in 90 degree mode. The design is 90-degree routable and a proof-of-concept routing is provided.

**To Reproduce**
Steps to reproduce the behavior:
1. Open `test.kicad_pcb` in KiCAD.
2. File -> Export -> Specctra DSN as `test.dsn`.
3. Run `freerouting` via CLI as per the below.
4. File -> Importt -> Specctra Session and observe unrouted nets.

**Expected behavior**
I expected Freerouting to either route the nets or provide more information about why it gave up/quit.

**Screenshots**

Freerouting output:
[image]

Manually routed proof-of-concept:
[image]

**Environment**
 - OS: Ubuntu 24.04 x86_64
 - EDA: KiCad 8.0.7
 - Freerouting version: 2.0.2-SNAPSHOT revision 5b1c365

Note: since Freerouting doesn't appear to have a way to force 90 degree routing via the CLI (as opposed to the GUI), I had to modify `trace_angle_restriction` below to force it to always be `AngleRestriction.NINETY_DEGREE`.

https://github.com/freerouting/freerouting/blob/5b1c365a09b27ba8c4ef592641a6979fc0f70de9/src/main/java/app/freerouting/rules/BoardRules.java#L38

**Command line arguments**
```shell
freerouting -ll 7 --gui.enabled=false --router.allowed_via_types=false --router.max_passes=10000 --router.fanout_max_passes=10000 --router.default_undesired_direction_trace_cost=0.01 --router.default_preferred_direction_trace_cost=0.01 --router.start_ripup_costs=1 -oit 0.001 -de test.dsn -do test.ses
```

**Project files**
If applicable, include project files like .pro, .kicad_pcb, .dsn.
See https://github.com/edwardcwang/freerouting-90deg-bug

**Logs**
```
2025-02-06 19:10:55.910 [main] INFO  Freerouting v2.0.2-SNAPSHOT (build-date: 2025-01-24)
2025-02-06 19:10:57.236 [ForkJoinPool.commonPool-worker-1] INFO  New version available: v2.0.1
2025-02-06 19:10:57.340 [Thread-7] INFO  Auto-router pass #1 on board '1b575aaf9b0e794f2e9428200ff7fe38' was completed in 0.12 seconds
2025-02-06 19:10:57.389 [Thread-7] INFO  Auto-router pass #2 on board 'a458a7ea65d54434e2593994bd7e7469' was completed in 0.04 seconds
```

edwardcwang, 2025-03-05:
Any updates on this issue?

andrasfuchs, 2025-03-05:
Might be related to #367? I'll investigate it further this week.