freerouting/freerouting #415: Request for GUI Language Setting

pakutoma, 2025-02-10:
I am very grateful for this tool, as it has been incredibly helpful for my hobby of electronic work.

However, as a Japanese speaker and a user of a PC with Japanese settings, I was confused by the fact that much of the text on the GUI settings screen was incomprehensible.
For example, almost every instance of the word "via" was translated as "by way of". There were many other incomprehensible terms throughout.

The issue here is not the quality of the Japanese translation itself, but rather that the default language setting depends on the PC's language settings and cannot be changed within the GUI.

Off topic, but I have corrected some misleading translations on Crowdin, including the “via” issue.
If it is better to submit a pull request to GitHub, I will do so.

In any case, I believe that low-quality translations, especially of technical terms, are more detrimental than having no translation at all.
Therefore, I would be very grateful if you could consider adding a language setting option to the GUI.

Thank you for your time and consideration.

andrasfuchs, 2025-02-12:
Thank you for the kind words and your work, @pakutoma!

Could you be so kind to create a pull request as well? It would make my work easier, so I would really appreciate it.

I'm not 100% sure that the Crowdin files are in sync with the repo. This is my fault, and I will fix this soon. Sorry if that causes some extra work.

I agree, language selection on the GUI sounds like a good idea. As far as I remember the difficulty was that the GUI is not prepared for run-time language changes, so Freerouting would need to be restarted if someone changed the language on the GUI. Still, I think that would be better than nothing, so I will look into it in a few weeks.

L1uTongweiNewAccount, 2025-02-17:
Hey, guy, SChinese was be in the same situation.

I think the old translation was made by machine, so I fixed them as #307, then TChinese translation was also pulled by other people.

Anyway, thanks for freerouting's international!