freerouting/freerouting #420: Heavily Memory Leak of route optimizator

L1uTongweiNewAccount, 2025-02-14:
Maybe like issue #390 , I think ...?

When I am parsing board including 979 pins (not too many for some appalications), It does well in routing.

But when I set optimize by option, I got `java.lang.OutOfMemoryError: Java heap space`.

After that, I set `-Xmx20000M` &`-mt 1`  in command-line, I have 32G of RAM, it also throw OOM execption.

I am sure about a heavily memory leak of it, Maybe about memory management mechanism in Java Virtual Machine.

I am sure that you can review this bug by a big board.

L1uTongweiNewAccount, 2025-02-17:
@andrasfuchs Don't worry, I will find a way to got the memory information.

I am suspicion about that, ..., if you always allocate memory, why not trigger the force garbage collection? I don't know about Java GC, ...

Anyway, I will try to run this another hours, and try to dump memory information by `jmap`

`jmap` is a tool that can dump the memory of freerouting. It will be a big file, about 8GB.

Then, open it by Memory Analyzer. I can analysis the memory usage.

Anyway, it needs time to find the problem behind this, so give me some time.

I will try to dump in the evening (Beijing Time)

L1uTongweiNewAccount, 2025-02-18:
From the Java VM's principle to say, you are handing the reference of all the objects for a long time (I think it's forever).

This is what google tell me. I'm downloading the analysis tools, so the details will be later.

L1uTongweiNewAccount, 2025-02-18:
@andrasfuchs OK, I got the status of memory usage.

MAT said that just 770KB of objects are reachable, that means we cannot get what code allocate the almost memory.

But I still got the unreachable objects lists, like this:

[expoeted_heap.txt](https://github.com/user-attachments/files/18843124/expoeted_heap.txt)

Also, I mentioned a special information, when I start the optimizer, although with signle thread, I still got 100% CPU.

In older version, this is unpossible, so I think you spent your most of time to allocate objects.

This is the output of taskmgr, for about 4 minutes.

[image]

That means if you start to debug, you can easily find the most be executed code is the key of memory leak.

andrasfuchs, 2025-02-18:
Excellent work, thank you!

I'll look into this using [visualvm](https://visualvm.github.io/) too, hoping that I can find that places where we wastefully allocate objects.

Do I understand it correctly that it happens only with the optimizer/post-router (and not with the router)?

terrancewong, 2025-05-20:
[image]

And 2.1 sometimes stuck forever for some easy connections, (pins on right column are all same net)

L1uTongweiNewAccount, 2025-09-17:
> > [@andrasfuchs](https://github.com/andrasfuchs) I found that in router, for normal board design, it will waste 3GB.
> > So I can infer that that exists memory leak with the router, but it's not infititly memory leak, so it can done the job in bigger heap space.
> > For Post-router, I don't know, it's just a easy process, so maybe we don't care.
> > Anyway, there must exists memory leak in routing process, but it's less neccessary than optimizer.
> > In fact, although you have 100PB heap size, it will be all-allocated in optimizer.
> 
> Can you test with the latest snapshot release? One potential memory leak source has been fixed and merged into main.

Sorry,  I have not too much time to do that.

Anyway, you can re-test it with testsuit internal (/tests), try to alloc it with 8GB, and run `jmap` to export memory, run `MAT` to analyize it.

If you haven't see a lot of memory leak, it will be ok.

andrasfuchs, 2026-01-17:
I'm working on this issue ATM.

The maximum heap memory usage for the first 5 passes were only 944 MBs on the 11 threads combined, and it was fairly stable during the whole process. 

```
2026-01-17 14:01:19.772 INFO   [139C12\D355A1] Auto-router session completed: started with 1125 unrouted nets, completed in 995.04 seconds, final score: 881.48 (413 unrouted, 0 violations), using 973.34 total CPU seconds, 446.38 GB total allocated, and 944 MB peak heap usage.
```

I think we don't have the memory leak anymore, but the router performance still does not yet meet my standards.

I'll keep working on it, but the good news is that it is very likely that the next release will not suffer from this memory leak issue.