freerouting/freerouting #425: Roadmap to version 2.1

andrasfuchs, 2025-02-18:
There are [a few issues already assigned to the 2.1 milestone](https://github.com/freerouting/freerouting/issues?q=is%3Aissue+milestone%3A2.1+) and there are a few steps that I plan to complete before the next release.

I copy them here from my personal notes for transparency, but I can't give deadlines since I have very little time to work on Freerouting. 
As always, any help is welcome!

- [ ] GUI
  - [ ] fix Japanese translation (review the PR done by )
- [ ] Backend
  - [ ] Investigate the memory leak/endless loop that prevents the router to finish big boards
  - [ ] Allow the CLI and the API to enable the fanout and the optimizer steps
  - [ ] Calculate the missing statistics of boards
  - [ ] Implement a scoring system that ranks the routing jobs/branches
- [ ] API
  - [x] Add a new streaming endpoint for progress updates during routing
  - [ ] Implement the /logs endpoint to show log entries associated with a job
- [ ] Docker
  - [ ] Review the PR for GitHub Docker registry
- [ ] Logging
  - [ ] Add the ability to assign log entries to sessions or jobs
  - [ ] Remove the thread name from the freerouting.log file entries

andrew64k, 2025-02-18:
Routing failed #367 is a rather important issue.

jsiddall, 2025-02-19:
> Routing failed [#367](https://github.com/freerouting/freerouting/issues/367) is a rather important issue.

Agree, this is my biggest issue

adamNewell, 2025-02-20:
I love the tool! I don't know if it's a KiCad side problem, but it would solve a huge UX issue for me if the tool remembered my settings from prior runs. I could also be ignorant of a workaround here...

andrasfuchs, 2025-02-20:
@andrew64k @jsiddall I agree, that would be important too. I've just added that issue to the roadmap.

andrasfuchs, 2025-02-20:
@adamNewell You might be able to keep your settings by modifying your `freerouting.json` file in your `%temp%` or `%temp%\freerouting` directory. Modifying the that file would change the default values, so you don't need to change them all the time.

Board-specific permanent settings were supported in the past, but I don't think they are working now, and I don't plan to implement them again with our new JSON format.

andrasfuchs, 2025-03-11:
@rozrabiak Although I feel it too that Java is not ideal, rewriting the whole codebase to any other language would be an enormous undertaking, so it will not happen. At least not done by me :) 

We have a wonderful [public API](https://api.freerouting.app/v1/system/status) though, so if you would like to use Freerouting from Python, you can do so fairly easily. You need to [apply for API usage at the website first](https://www.freerouting.app/), and we have [an API documentation](https://github.com/freerouting/freerouting/blob/master/docs/API_v1.md) to start with.

Could you explain in detail (preferably in new issues) what do you exactly mean by AI support and auto-replacement of footprints?

tapegoji, 2025-03-30:
I don't think generic AI would do anything here. However, a reinforcement model might add strength. This is a big task and not sure it can be done even in a year. What do you think Andras?

andrasfuchs, 2025-03-30:
I would support the idea of reinforcement learning in general, there are some good options like [Deep Q-leaning](https://www.geeksforgeeks.org/deep-q-learning/) and our new scoring system in the next release could provide a fantastic foundation for implementing such techniques.

That being said the current community donations don't justify the significant time investment required for further development, given my existing professional commitments. I will not abandon the project completely, but I will focus on critical fixes only after the v2.1 release. 

This could change in the future if a company/foundation would offered to sponsor the project's continued development. I'd be eager to connect with any companies that could benefit from a higher quality auto-router and would be open to keep the project open source and free for others.

tapegoji, 2025-03-30:
I sent you a LinkedIn request. Let's have a chat about this. I have some ideas.

closed 2025-04-12 by andrasfuchs