freerouting/freerouting #442: issue with "_" when reading dsn files or .rules file

tapegoji, 2025-03-30:
**Describe the bug**
When reading .dsn or .rules file if there is an underscore, "_" in the name it mistakes it as a an operation. for example:
Below you can see that in the first line you are defining clearance between hv and default. 
In the second line I define between hv and kicad_default net class but it does not process it. 

(rules PCB vdiv_placed.dsn
  (rule
      (clearance 500 (type hv_default))
      (clearance 400 (type hv_kicad_default))
  )
)

**To Reproduce**
simply run one of the examples. in the test for example that has _  in the name.

tapegoji, 2025-03-30:
If you tell me which file to look at, I might be able to fix it. Any hint is appreciated.

tapegoji, 2025-03-31:
I think I fixed it. Look at this my PR and let me know if you want me to change anything

tapegoji, 2025-04-01:
I wanted to ask you a question that I forgot to ask. For clearance class separation, do you use '-' or '_'. I see that in the code you have a constant that is using '-' however, In some other parts you are using '_'. Which one should I go with?

andrasfuchs, 2025-04-02:
Yes, it seems like a little problematic, because depending on the tool that generated the .rules file, it can be both, I think.

You mentioned that you use [FreeCAD](https://www.freecad.org/), right? Could you send me a few .rules files that you have and I can test against?

I also uploaded [a test file for this issue](https://github.com/freerouting/freerouting/blob/master/tests/Issue442-clearance_type_tests.rules), just to see what combinations are possible for defining the clearance rules.

For example, we should be able to process this file even if the formatting is mixed, and there are duplicate definitions:

```
  (rule
    (width 456.7)
    (clear 153.4)
    (clear 112.3 (type "smd_to_turn_gap"))
    (clear 145.6 (type smd_to_turn_gap))
    (clear 57.8 (type smd_smd))	
    (clear 199.1 (type "default"-"1A EXTERNAL 1oz"))
    (clear 192.3 (type "default"-"2.5A EXTERNAL"))
    (clear 1456.7 (type "default"-"3,5A EXT HIGH VOLTAGE"))

```

tapegoji, 2025-04-02:
I am able to process these rules listed here. As you can see in the screenshot clearance matrix has been assembled correctly. 

`    (rule
      (width 200)
      (clearance 200)
      (clearance 50 (type smd_smd))
      (clearance 100.0 (type "GND"-"VOUT"))
      (clearance 150.0 (type "GND"-"R3_P"))
      (clearance 200.0 (type "R3_P"-"VOUT"))
      (clearance 600.0 (type VIN-VOUT))
      (clearance 700.0 (type VIN-"R3_P"))
      (clearance 800.0 (type VIN-"External 5V"))
      (clearance 900.0 (type VIN-"External 5V"))
      (clearance 1000.0 (type "External 5V"-VOUT))
    )`

[image]

closed 2025-04-12 by andrasfuchs