freerouting/freerouting #445: Many clearance violations after autorouting

Gifford47, 2025-04-02:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. load attached .dsn file (4 Layer)
2. Start Freerouting autorouting
3. Importing .ses file back to easyeda Pro >v2.2.36.7
4. Run DRC (rule check)

**Expected behavior**
No clearance violations after DRC

**Screenshots**

[image]

[image]

[image]

**Environment**
 - OS: Windows 11, 64-bit
 - Freerouting version: v2.0.1
 - EDA: EasyEDA PRO v2.2.36.7

**Command line arguments**
GUI application used

**Project files**

EasyEDA Pro Multilayer DesignRule:
[Config_DesignRule_Board_2025-04-02.json](https://github.com/user-attachments/files/19564797/Config_DesignRule_SmartNet.Rail.Board_2025-04-02.json)

specctra .dsn file:
[test.zip](https://github.com/user-attachments/files/19565122/test.zip)

**Logs**
Include at least the most recent lines of your log files or console output.

**Additional context**
It seems, that the clearance rules are ignored or others are loaded while routing. The clearance matrix is also different from EDA rules. The output ses file causes many violations after importing to eda.

Please look at the .dsn file. I'm not quite sure whether the problem is with the export of the dsn file in the eda.

pierremolinaro, 2025-04-02:
> Gifford47
>  left a comment
> (freerouting/freerouting#445)
> @andrasfuchs Do you have any plan what's wrong with the .dsn file or the rules?

Hello,

I'm using an older version of FreeRouter and I've encountered the clearance violation problem.

I solved it by adding a “clearance” setting to the default rule.

For the test.dsn file of #445, change line 18 from

    (rule(width 1.01))

To

    (rule(width 1.01)(clearance 0.254))

I don't know if this modification is still valid for the current version of FreeRouter.

Hope that helps,

Pierre Molinaro

Gifford47, 2025-04-02:
ok, thanks! now I know how the syntax works!
could you post your dsn file please? Or could you explain how the resolution works?
f.e. the following is present in my dsn file:
```
  (resolution mil 1000)
```
all values are in mils and are internally divided by 1000, right?
what means that for ```(rule(clear 0.41))``` ?

Gifford47, 2025-04-02:
unfortunately ```(rule(width 1.01)(clearance 0.254))``` makes no difference

andrasfuchs, 2025-04-02:
@Gifford47 thanks for report and your efforts for helping the bug hunt!

I haven't looked into it yet, but I will check it before the next release. It looks like an important fix, but I can't promise anything at the moment, we'll see how much free time I have in the next few weeks.

In the meantime please try to provide me as many examples and files as you can, so that I can test this and confirm the fix more quickly.

Thanks for you patience!

pierremolinaro, 2025-04-02:
Your file uses mil resolution, so 0.254 means 0.254 mil, perhaps it is too small ?

My file uses mm resolution, my clearance is 10 mil, so the value is given by  10 mil x 25.4 mm/in x 1 in / 1000 mil =  0.254 mm

I think your clearance is 0.41 mil, try :

    (rule(width 1.01)(clearance 0.41))

Here is my design:

￼

Pierre

Gifford47, 2025-04-04:
ok, problem solved. the issue is/was the exported dsn file from eda. there are to less class and rule definitions (f.e. pin_pin).
in short: I've made some python scripts which fix the dsn file.
If someone is interested, the scripts are attached.
You need the dsn and the design rule file as json file in the same folder. Both must be in the same unit (mil, mm, inch, etc,) format!

[fix easyeda dsn.zip](https://github.com/user-attachments/files/19604296/fix.easyeda.dsn.zip)

closed 2025-04-04 by Gifford47

vaelen, 2025-04-05:
@Gifford47 , did you submit a bug report to EasyEDA?

Gifford47, 2025-04-05:
Did it two times for two different issues but still no issue was solved.
Please send as many requests as you can to easyeda because dsn exporter is very buggy and has a very simple and basic structure! It definitely needs to be improved !