freerouting/freerouting #448: Not wotking on Nixos

Ben-Miller0, 2025-04-08:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on the free routing icon in the pcb editor in kicad

**Expected behavior**
the program to run

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - OS: Nixos 24.11
 - Freerouting version: 2.0.1
 - EDA: KiCad

**Logs**
Traceback (most recent call last):

  File "/home/benmiller/.local/share/kicad/8.99/3rdparty/plugins/app_freerouting_kicad-plugin/plugin.py", line 354, in Run
    self.RunSteps()

  File "/home/benmiller/.local/share/kicad/8.99/3rdparty/plugins/app_freerouting_kicad-plugin/plugin.py", line 340, in RunSteps
    if not self.RunRouter() :
           ^^^^^^^^^^^^^^^^

  File "/home/benmiller/.local/share/kicad/8.99/3rdparty/plugins/app_freerouting_kicad-plugin/plugin.py", line 311, in RunRouter
    invoker.show_error()

  File "/home/benmiller/.local/share/kicad/8.99/3rdparty/plugins/app_freerouting_kicad-plugin/plugin.py", line 570, in show_error
    command = " ".join(self.command)
              ^^^^^^^^^^^^^^^^^^^^^^

TypeError: sequence item 4: expected str instance, PosixPath found

giomf, 2025-04-17:
How did you manage to get it running in the first place?
Even after installing jre21, kiCad always says that jre21 is missing when hitting the freerouting button in pcb-designer.

closed 2025-08-17 by Ben-Miller0

jVallant, 2025-11-12:
For others with the same problem, changing the jre package from "jre_minimal" to "temurin-jre-bin" solved the issue for me.

jre_minimal does not seem to have swing. Directly running gives the following: 
```
> java -jar freerouting-2.1.0.jar
Error: Unable to initialize main class app.freerouting.Freerouting
Caused by: java.lang.NoClassDefFoundError: javax/swing/UnsupportedLookAndFeelException```