freerouting/freerouting #461: KiCad: Freerouting 2.1.0 is broken, can't auto-route same PCB as 2.0.1, need to revert

0ddjob, 2025-04-17:
**Problem**
I used Freerouting 2.0.1 to auto-route a PCB successfully.  I then noted that 2.1.0 was available so installed that and re-tried.  The same PCB can no longer be successfully auto-routed, leaving 13 to 24 (it varies on each run) un-routed. 
Auto-routing is now unfortunately broken.

**To Reproduce**
Steps to reproduce the behavior:
1. Load attached Kicad PCB
2. Try to auto-route with Freerouting 2.1.0

**Expected behavior**
With Freerouting 2.0.1 the PCB is successfully auto-routed quite quickly.
With Freerouting 2.1.0 the PCB takes a lot longer to auto-route and does not complete.

**Screenshots**
[image]

**Environment**
 - OS: macOS 15.3.2 (24D81)
 - Freerouting version: 2.1.0
 - EDA: KiCad 9.0.0

**Command line arguments**
Ran from KiCad -> Tools -> External Plugins

**Project files**
[Nafanya_Keyboard_Kicad.zip](https://github.com/user-attachments/files/19786299/Nafanya_Keyboard_Kicad.zip)

0ddjob, 2025-04-17:
Reverted to Freerouting v2.0.1 and PCB could be routed again with no issue.

[image]

CaptainDangeax, 2025-05-05:
Same issue when upgrading from 1.4.4 to 2.0.1. Just to follow