freerouting/freerouting #531: Freerouter does not always respect clearence limits when routing

softfoot, 2025-08-20:
**Describe the bug**
99% of the time it does honour the clearence settings, but it does not always do so when routing near :-
    A. Near the "board cut" outline.
    B. Near the little "wiggles" where a track or differential pair has been "tuned" for length or "skew".

**To Reproduce**
1. Using a fairly complex board desig wheru FR is forced to consider routing near the card edge etc.
2. Allow it to route with the default settings.
3. Run DRC and look for errors/warnings.

**Expected behavior**
It should honour the clearance settings.

**Environment**
 - OS: Windows 10 64-bit]
 - Freerouting version: 2.1.0
 - EDA: KiCad 9.04

**Command line arguments**
None - run from KiCad FR tool.

**Project files**
Sorry, I am not permitted to.

**Logs**
It is fairly consistent in doing this, if youtell me how the collect the logs I'll post them.

cgreening, 2025-09-29:
Looks related to my issue: https://github.com/freerouting/freerouting/issues/558

andrasfuchs, 2026-01-29:
Thank you Dave for the report, it sounds like an important issue.

I'm afraid I will be not able to fix this without a simple test file that demonstrates the issue.

I can see that you are not allowed to share the whole board, but can you give me a simplified version that still lets me test the potential fix? KiCad project files would be better, but a plain DSN file would be sufficient to start.

I would greatly appreciate it. Thank you!