freerouting/freerouting #558: Copper to edge clearance doesn't seem to be respected

cgreening, 2025-09-24:
**Describe the bug**
The autorouting is successful, but when I run a DRC. In my board setup I've got it set to 0.5mm.

But, when I run a DRC on the autorouted results, I get clearance violations.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up a board with a Copper to edge clearance set to 0.5mm
2. Run freerouting with a board that will result in tracks routed close to the edge of the board
3. DRC results in failures for the board edge clearance.

**Expected behavior**
DRC should pass.

**Screenshots**
[image]

**Environment**
 - OS: Mac OS 15.6.1 (24G90)
 - Freerouting version: 2.1.0
 - EDA: KiCad 9

**Command line arguments**
Launched from the KiCad button.

**Project files**
If applicable, include project files like .pro, .kicad_pcb, .dsn.

[dev-board-autoroute-demo.zip](https://github.com/user-attachments/files/22517968/dev-board-autoroute-demo.zip)

andrasfuchs, 2025-11-14:
@gemini-cli Can you tell me what classes of the codebase could potentially be responsible for this problem?

andrasfuchs, 2025-11-14:
@gemini-assistant Can you tell me what classes of the codebase could potentially be responsible for this problem?

andrasfuchs, 2025-11-14:
@gemini-cli help me debug this error

andrasfuchs, 2025-11-15:
@gemini-cli Help me debug this issue

andrasfuchs, 2026-09-17:
Closing as resolved in Freerouting.

As investigated, KiCad's Specctra DSN exporter does not write the custom copper-to-edge clearance value into the boundary scope of the `.dsn` file (tracked upstream in KiCad [GitLab #24077](https://gitlab.com/kicad/code/kicad/-/work_items/24077)).

To address this on the Freerouting side:
1. PR #567 fixed non-zero layer edge clearance handling.
2. Added the configuration parameter `--router.copper_to_edge_clearance_um` (e.g. `--router.copper_to_edge_clearance_um=500` for 0.5 mm clearance), with a default of 250 µm (0.25 mm).

This allows users to match KiCad's board setup clearance rules cleanly from both the CLI and configuration files until upstream KiCad emits the clearance class in DSN exports.

closed 2026-09-17 by andrasfuchs